first95/FRC2016 #4: Develop Aiming Equations

setupminimal, 2016-01-15:
Figuring out our position on the field, the force with witch to launch, and at what angles.

jwalthour, 2016-01-15:
$0.02: The first step in this task is figuring out the right inputs and outputs to these equations.  Bear in mind, we currently do not have plans for the robot to know its location on the field (as far as I know).  What we do have is plans to have a working camera, and the knowledge of where a high goal is relative to the robot.  From those inputs, it would be useful to know: 1) by what angle does the robot need to turn in azimuth? 2) at what elevation angle must the ball be fired? 3) at what speed must the ball be fired?

You'll also notice that, when firing along a parabolic arc, the arc can intersect the goal at many points.  This means your algebra will come up with N equations in N+1 unknowns, representing a family of solutions, all of which are valid.  You may find it easiest to develop equations that place the ball in the goal opening at its apogee.  This means the ball will pass through the goal when the trajectory is at its flattest, which gives us room for error in range.

jwalthour, 2016-01-16:
Where:
V0y, V0x are the (x,y) components of the initial velocity of the ball, in inches per second
theta, m are the polar form of <V0x,V0y>
h is the height of the shooter
g is the acceleration due to gravity, in inches per second squared
X is the range between the robot and the base of the tower
97" is the height of the center of the high goal
t is the time required for the ball to reach apogee

V0y=sqrt(2gh)
t=V0y/g
V0x=X/t
theta = arctan(V0y/V0x)
m=sqrt(V0x^2+V0y^2)

jwalthour, 2016-01-16:
[image]

closed 2016-01-16 by jwalthour