first95/FRC2016 #6: Get it so we can read attitude data from the ArduPilot

jwalthour, 2016-01-16:


dcavanau, 2016-01-16:
The wpilib plugin adds some classpath variables to your workspace, so if you've deleted yours or have created a new one, you'll need to add them again.

If you create a new Robot Java Project the wpilib plugin will automatically add the classpath variables. Otherwise, follow the steps below to add them manually:
Open the Preferences window and navigate to: Java > Build Path > Classpath Variables
Add the following classpath variables, substituting <User> with your username:
networktables: C:/Users/<User>/wpilib/java/current/lib/NetworkTables.jar
networktables.sources: C:/Users/<User>/wpilib/java/current/lib/NetworkTables-sources.jar
wpilib: C:/Users/<User>/wpilib/java/current/lib/WPILib.jar
wpilib.sources: C:/Users/<User>/wpilib/java/current/lib/WPILib-sources.jar

jwalthour, 2016-02-07:
Completed in https://github.com/first95/FRC2016/commit/8c30446a4af59c2e3b3f537bdb9dc7a5ba96b518

closed 2016-02-07 by jwalthour