first95/FRC2016 #12: Knowledge About PIDF

setupminimal, 2016-02-06:
First: The F term accounts for 95% of the adjustment. Set the F term about as though you were in open loop mode.

The other three terms deal with eccentricities of the motor, etc.

Also: Voltage output varies with battery voltage, so if you correct for the battery voltage, you get better performance.

setupminimal, 2016-02-06:
Done for now.

closed 2016-02-06 by setupminimal

jwalthour, 2016-02-06:
My phrasing of @setupminimal 's first comment:
Roughly speaking, if Vout is the effective voltage sent to the motor, s is the setpoint value, and e is the "error" between the setpoint and the current speed:
Vout=F*s + P*e + I * (integral of e with respect to time) + D * (de/dt)
Thus, F does not refer to the previous iteration of the control loop, but rather refers to a shortcut around the normal control loop.